connortechnology/ZoneMinder #9: "Bogus number of lines return from query, 0 returned for query select"

shaun-ba, 2017-03-16:
Getting these on a fresh install, don't get them with exactly the same process on master (1.30.0)

`2017-03-16 15:14:30.703116	zms		23047	ERR	Bogus number of lines return from query, 0 returned for query select Id, Name, ServerId, StorageId, Type, Function+0, Enabled, LinkedMonitors, Device, Channel, Format, V4LMultiBuffer, V4LCapturesPerFrame, Protocol, Method, Host, Port, Path, Options, User, Pass, Width, Height, Colours, Palette, Orientation+0, Deinterlacing, RTSPDescribe, SaveJPEGs, VideoWriter, EncoderParameters, RecordAudio, Brightness, Contrast, Hue, Colour, EventPrefix, LabelFormat, LabelX, LabelY, LabelSize, ImageBufferCount, WarmupCount, PreEventCount, PostEventCount, StreamReplayBuffer, AlarmFrameCount, SectionLength, FrameSkip, MotionFrameSkip, AnalysisFPS, AnalysisUpdateDelay, MaxFPS, AlarmMaxFPS, FPSReportInterval, RefBlendPerc, AlarmRefBlendPerc, TrackMotion, SignalCheckColour, Exif from Monitors where Id = 0.	zm_db.cpp	125`

`2017-03-16 15:14:30.682988	web_js		22914	ERR	getStreamCmdResponse stream error: Socket /var/run/zm/zms-261302s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()	?view=watch`

`2017-03-16 15:14:30.670600	web_php		22914	ERR	Socket /var/run/zm/zms-261302s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2005`

connortechnology, 2017-03-20:
please try the latest packages.

shaun-ba, 2017-03-22:
On install your scripts tried to start ZM and this was the result:

Mar 22 08:10:25 zm-vid zmpkg.pl[24116]: Compilation failed in require at /usr/bin/zmdc.pl line 65.
Mar 22 08:10:25 zm-vid zmpkg.pl[24116]: BEGIN failed--compilation aborted at /usr/bin/zmdc.pl line 65.
Mar 22 08:10:25 zm-vid sudo[24243]: pam_unix(sudo:session): session closed for user www-data
Mar 22 08:10:25 zm-vid zmpkg[24116]: ERR [Unable to run "sudo -u www-data /usr/bin/zmdc.pl start zma -m 5", output is ""]
Mar 22 08:10:25 zm-vid zmpkg.pl[24116]: DBD::mysql::st execute failed: MySQL server has gone away at /usr/share/perl5/ZoneMinder/Logger.pm l
Mar 22 08:10:25 zm-vid zmpkg[24116]: FAT [Can't execute log entry 'insert into Logs ( TimeKey, Component, Pid, Level, Code, Message, File, L
Mar 22 08:10:25 zm-vid systemd[1]: zoneminder.service: Control process exited, code=exited status=255
Mar 22 08:10:25 zm-vid systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.
Mar 22 08:10:25 zm-vid systemd[1]: zoneminder.service: Unit entered failed state.
Mar 22 08:10:25 zm-vid systemd[1]: zoneminder.service: Failed with result 'exit-code'.

connortechnology, 2017-03-22:
Might want to check your mysql logs.  How many cameras are you using?

shaun-ba, 2017-03-22:
I'll check, but i didn't get this in the previous package.

shaun-ba, 2017-03-22:
So, on starting the new package this is shown in mysql logs, note again that i didn't touch anything else just updated.

`170322 13:32:44 mysqld_safe Number of processes running now: 0
170322 13:32:44 mysqld_safe mysqld restarted
170322 13:32:44 [Note] /usr/sbin/mysqld (mysqld 10.0.29-MariaDB-0ubuntu0.16.04.1) starting as process 25221 ...
170322 13:32:44 [Note] InnoDB: Using mutexes to ref count buffer pool pages
170322 13:32:44 [Note] InnoDB: The InnoDB memory heap is disabled
170322 13:32:44 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
170322 13:32:44 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
170322 13:32:44 [Note] InnoDB: Compressed tables use zlib 1.2.8
170322 13:32:44 [Note] InnoDB: Using Linux native AIO
170322 13:32:44 [Note] InnoDB: Using CPU crc32 instructions
170322 13:32:44 [Note] InnoDB: Initializing buffer pool, size = 128.0M
170322 13:32:44 [Note] InnoDB: Completed initialization of buffer pool
170322 13:32:44 [Note] InnoDB: Highest supported file format is Barracuda.
170322 13:32:44 [Note] InnoDB: Log scan progressed past the checkpoint lsn 2948786
170322 13:32:44 [Note] InnoDB: Database was not shutdown normally!
170322 13:32:44 [Note] InnoDB: Starting crash recovery.
170322 13:32:44 [Note] InnoDB: Reading tablespace information from the .ibd files...
170322 13:32:44 [Note] InnoDB: Restoring possible half-written data pages 
170322 13:32:44 [Note] InnoDB: from the doublewrite buffer...
InnoDB: Doing recovery: scanned up to log sequence number 2965472
170322 13:32:44 [Note] InnoDB: Starting an apply batch of log records to the database...
InnoDB: Progress in percent: 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 
InnoDB: Apply batch completed
170322 13:32:45 [Note] InnoDB: 128 rollback segment(s) are active.
170322 13:32:45 [Note] InnoDB: Waiting for purge to start
170322 13:32:45 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.34-79.1 started; log sequence number 2965472
170322 13:32:45 [Note] Plugin 'FEEDBACK' is disabled.
170322 13:32:45 [Note] Server socket created on IP: '127.0.0.1'.
170322 13:32:45 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.0.29-MariaDB-0ubuntu0.16.04.1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Ubuntu 16.04`

shaun-ba, 2018-01-30:
fixed in 1.31.29

closed 2018-01-30 by shaun-ba