connortechnology/ZoneMinder #13: Events not recorded at "real" speed

shaun-ba, 2017-03-25:
My events when played back are at a much faster speed than real time. The downloaded mp4 is the same.

Let me know if there is some more information i can provide

```
Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : Main@L4.1
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 2 frames
Format settings, GOP                     : M=1, N=10
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 10s 700ms
Bit rate mode                            : Variable
Bit rate                                 : 15.8 Mbps
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 20.000 fps
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.381
Stream size                              : 20.2 MiB (100%)
Encoded date                             : UTC 1904-01-01 00:00:00
Tagged date                              : UTC 1904-01-01 00:00:00
Color range                              : Limited
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709
```

bhaal2, 2017-04-01:
This might be because the framerate has not been saved in the mp4 file when the file is closed off, setting a framerate when playing the video is a quick but annoying fix.

shaun-ba, 2018-01-30:
fixed in 1.31.29

closed 2018-01-30 by shaun-ba