connortechnology/ZoneMinder #16: Log cannot be displayed

Rayn0r, 2017-06-14:
I'm currently running ZM in  Version 1.31.0 from git clone `617cdd5fd385cc0dfd75b8aa56980ef42f492df7`
on Ubuntu 16.04

For quite some time I have not been able to view the log messages over the web interface. Although all logs reside under /var/log/zm . Upon clicking the Log link on the ZM interface, the file web_php.log will be flooded with 7000-8000 lines of json formated data followed by the error message 

> unexpected error 5] at /usr/share/zoneminder/www/includes/functions.php line 1938.

Could the problem be related to a comma at the end of every json element? As far as I know it is not supposed to be there.

Find the log excerpt here: [web_php.log](https://github.com/connortechnology/ZoneMinder/files/1074336/web_php.log.txt)

connortechnology, 2017-06-14:
Are you actually storing logs in the db?  That is what the log view shows.  Not the contents of /var/log/zm/

The logs view works for me using current master.  

Line 1938 of functions.php has to do with X10 commands.  Are you using the X10 code?

Your web_php.log contains bad characters (non-utf-8) which cannot be decoded. This is the core of the problem.  Ideally zm would strip them out from the logs before sending.  I may have to look into that.

Rayn0r, 2017-06-14:
> Are you actually storing logs in the db?
I think so. The link used to work. I have not used it for a few weeks. So I do not know when ist stopped working.
I recall that it has failed before for a day or two and then suddenly worked again. During that time I had not installed an update for ZM.

> Line 1938 of functions.php has to do with X10 commands. Are you using the X10 code?
Not that I am aware of. I am using a POE Camera. X10 is a powerline protocol, right?

Rayn0r, 2018-02-18:
Today was the first time I was able to see the Log ever since I created the issue. 
I'm not saying it's gone for good, but I will keep a closer look...

Rayn0r, 2018-05-04:
I'd say its safe to close the issue. Displaying the log has been working for the last couple of months.

closed 2018-05-04 by Rayn0r

connortechnology, 2018-05-04:
The log is unusable if you have a lot of log entries. It works great if you only have a few.