connortechnology/ZoneMinder #17: ffmpeg using all CPUs upon opening Events

Rayn0r, 2017-06-18:
I'm running latest HEAD `75c11431e622ccd274d51812c3e313b0cb204142` on my machine.

Upon opening the Event-View it takes 3-4 minutes before the content shows up. When opening the view Apache starts multiple ffmpegs. The command line looks like this:

`ffmpeg -ss 634.84 -i /var/cache/zoneminder/events/1/17/06/18/22/00/00/6206-video.mp4 -frames:v 1 /var/cache/zoneminder/events/1/17/06/18/22/00/00/15848-capture.jpg`

Running this outside zoneminder with `time` shows:
```
real    1m31.399s
user    3m33.084s
sys     0m6.960s
```

`ffprobe /var/cache/zoneminder/events/1/17/06/18/22/00/00/6206-video.mp4`
shows this:

```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/var/cache/zoneminder/events/1/17/06/18/22/00/00/6206-video.mp4':
  Metadata:
    major_brand     : iso5
    minor_version   : 512
    compatible_brands: iso6mp41
    title           : Zoneminder Security Recording
    encoder         : Lavf56.40.101
  Duration: 00:15:00.36, start: 0.000000, bitrate: 5005 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuvj420p(pc, bt709), 1920x1080, 5002 kb/s, 24.99 fps, 25 tbr, 90k tbn, 50 tbc (default)
    Metadata:
      handler_name    : VideoHandler
```

This is a 15 min file. Since ffmpeg has to decode 10 mins of the video to capture a single frame it takes a while.

Jumping ~70% into the video is probably not a good idea. Since we know the event length, isn't it best to only jump a few seconds into the recording, but no further than the event length?

Rayn0r, 2017-06-19:
I found the [code](https://github.com/connortechnology/ZoneMinder/blob/storageareas/web/includes/Event.php#L171) section responsible this, but it's been there since Jan 2nd. If I understood correctly, then a thumbnail is created from the frame with the highest score. Is there a way to postpone this until the page was loaded, or to create the image after the event was closed?
In commit 4889f3d0e8866171ca47dfc79f8f2f6e81dd8663 you reworked the logic behind this. Could this be the culprit?

connortechnology, 2017-06-19:
Under the storage tab, have it save a snapshot jpg instead of saving no jpegs.  Then ZM won't have to try to generate one.

On my system, ffmpeg generates this image almost instantly, but then my mp4's are not 15 minutes long.  

I think you are right though.  For long events we should limit how far in thumbnail generation will go.

connortechnology, 2017-10-11:
This might be slightly mitigated on latest builds.  We were ignoring the existence of  a snapshot image.

Rayn0r, 2017-10-14:
After updating to latest git, event pages load almost instantly, when snapshot image is enabled.
ffmpeg is not spawned anymore upon opening a page.
Thank you very much.

closed 2017-10-14 by Rayn0r