connortechnology/ZoneMinder #23: Montage review timeline does not reorder on resize (captions do)

MalakymR, 2017-10-04:
zm 1.31.6 storageareas

Under montage review, when resizing a view, the timeline captions reorder - but the timeline itself doesn't.

To demonstrate this I record with 2 cameras. Stop one for a short period of time before restarting it, this shows a clear gap in the timeline so we know which camera is which. Go to montage review and show the last hour. Resize camera 2 and the caption for camera 2 moves up, but the timeline itself does not change - so the captions are now on the wrong timeline.

This had occured in a few versions prior, including zoneminder/zoneminder 1.31.4.

Occurs in Firefox 56, 57, Chrome 59, Vivaldi 1.11 and IE11

[image]

MalakymR, 2017-10-06:
Is there a reason why the order of these timelines need to change? I can't think of a reason to do so, could simplify it otherwise.

connortechnology, 2017-10-06:
Agreed, I see no reason.

MalakymR, 2017-10-12:
Even if someone thinks the graphs should be reordered, it'd probably be better to remove that line until the work to do so is complete.

MalakymR, 2017-10-16:
Fixed by merge.

PR #36 (2017-10-15): MontageReview ordering
Round 2!

fixes #23 

This fixes captions reordering and graphs not by stopping anything from reordering.  This means the monitor views don't reorder on size either.  I have a fix that keeps monitor view reordering intact.  I think that's confusing because the monitor views don't match up with the graphs.  If you feel it's needed I think I could make everything reorder.
closed 2017-10-16 by MalakymR