connortechnology/ZoneMinder #25: Storageareas 1.31.7 - Multiselect ignored when deleting groups

MalakymR, 2017-10-05:
When selecting multiple groups to delete, only the first is deleted.

URL shows multiple gids for deletion /zm/index.php?view=groups&action=delete&gid=7&gid=9

Checked elsewhere, deleting cameras from the console, multiple selection/delete works okay.

MalakymR, 2017-10-09:
I've looked into this, and compared it against another multiselect operation (deleting cameras etc).

Groups deletion is not looping the array - this seemed strange at first because the parameter is using array() around the $_REQUEST so looked okay. But its only reading the first value. (Also its been a long time since I've done web work, not used to MVVM etc).

The camera deletion operation uses foreach() through the $_REQUEST to delete each... but still uses array() on a single value in the SQL delete statement, which I'm pretty sure is not needed. includes/actions.php Line 556.

`foreach( $_REQUEST['markMids'] as $markMid ) {`
`...`
`dbQuery( 'DELETE FROM Monitors WHERE Id = ?', array($markMid) );`
`dbQuery( 'DELETE FROM Zones WHERE MonitorId = ?', array($markMid) );`
`if ( ZM_OPT_X10 )`
`dbQuery( 'DELETE FROM TriggersX10 WHERE MonitorId=?', array($markMid) );`

I'm not going to touch markMid for now, but will PR in a bit for the others for review.

MalakymR, 2017-10-09:
I was wrong about not needing array() - I assume that dbQuery assumes the param is in an array regardless of values.

connortechnology, 2017-10-10:
I did something different in regards to this issue.  Thanks for your efforts though!  Please test what I've done.

MalakymR, 2017-10-10:
Fixed now.

When the popup window closes for groups, the parent page doesn't seem to refresh though.

closed 2017-10-17 by connortechnology