connortechnology/ZoneMinder #41: The video could not be loaded, either because the server or network failed, or the format is not supported

shaun-ba, 2017-10-23:
So i've trying this branch again but so far it isn't usable for me, I'll log some issues here.

The first is events cannot be played in browser, I received the message "The video could not be loaded, either because the server or network failed, or the format is not supported"

Here are some relevant logs:

2017-10-23 09:36:12.683932 | zms |   | 1260 | FAT | JPEGS not saved.zms is not capable of streaming jpegs from mp4 yet | zm_eventstream.cpp | 609
-- | -- | -- | -- | -- | -- | -- | --
2017-10-23 09:36:12.682657 | zms |   | 1260 | WAR | Unable to open ffmpeg_input /var/cache/zoneminder/events/3/17/10/23/08/27/55/2-video.mp4 | zm_eventstream.cpp | 220
2017-10-23 09:36:12.681766 | zms |   | 1260 | ERR | Could not open input file '/var/cache/zoneminder/events/3/17/10/23/08/27/55/2-video.mp4' (error 'No such file or directory') | zm_ffmpeg_input.cpp

MalakymR, 2017-10-23:
Does /var/cache... exist as its trying to say? What permissions does that folder have?

shaun-ba, 2017-10-23:
Yes the folder exists and it's owned by www-data. However, i notice that there is no video.mp4 there, so its not being created at all. Any idea what could cause that? these logs don't tell me that some dependency is maybe missing

shaun-ba, 2017-10-23:
New log

2017-10-23 09:53:25.581305 | zmc_m3 |   | 1586 | ERR | Backtrace 9: /usr/bin/zmc(_start+0x29) [0x56472db81ed9] | zm_signal.cpp | 102
-- | -- | -- | -- | -- | -- | -- | --
2017-10-23 09:53:25.580916 | zmc_m3 |   | 1586 | ERR | Backtrace 8: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f959dc2c830] | zm_signal.cpp | 102
2017-10-23 09:53:25.580519 | zmc_m3 |   | 1586 | ERR | Backtrace 7: /usr/bin/zmc(main+0x77e) [0x56472db80e5e] | zm_signal.cpp | 102
2017-10-23 09:53:25.580154 | zmc_m3 |   | 1586 | ERR | Backtrace 6: /usr/bin/zmc(_ZN7Monitor7CaptureEv+0x38c) [0x56472dbac4dc] | zm_signal.cpp | 102
2017-10-23 09:53:25.579737 | zmc_m3 |   | 1586 | ERR | Backtrace 5: /usr/bin/zmc(_ZN16RemoteCameraRtsp16CaptureAndRecordER5Image7timevalPc+0x6a8) [0x56472dbc9eb8] | zm_signal.cpp | 102
2017-10-23 09:53:25.579116 | zmc_m3 |   | 1586 | ERR | Backtrace 4: /usr/bin/zmc(_ZN10VideoStore21writeVideoFramePacketEP8AVPacket+0x241) [0x56472dc1bc41] | zm_signal.cpp | 102
2017-10-23 09:53:25.578750 | zmc_m3 |   | 1586 | ERR | Backtrace 3: /usr/lib/x86_64-linux-gnu/libavformat-ffmpeg.so.56(av_interleaved_write_frame+0x79) [0x7f95a06d2aa9] | zm_signal.cpp | 102
2017-10-23 09:53:25.578425 | zmc_m3 |   | 1586 | ERR | Backtrace 2: /usr/lib/x86_64-linux-gnu/libavformat-ffmpeg.so.56(+0xed769) [0x7f95a06d0769] | zm_signal.cpp | 102
2017-10-23 09:53:25.578043 | zmc_m3 |   | 1586 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f95a1e0c390] | zm_signal.cpp | 102
2017-10-23 09:53:25.576661 | zmc_m3 |   | 1586 | ERR | Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x78) [0x56472dbf21f8] | zm_signal.cpp | 102
2017-10-23 09:53:25.575644 | zmc_m3 |   | 1586 | ERR | Signal address is (nil), from 0x7f95a06d0769 | zm_signal.cpp | 81
2017-10-23 09:53:25.574681 | zmc_m3 |   | 1586 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
2017-10-23 09:53:25.495369 | zmc_m3 |   | 1586 | ERR | Can't open in codec!

connortechnology, 2017-10-23:
The important line here is the Can''t open in codec!  Meaning for some reason ffmpeg couldn't open the AAC encoder to encode audio.  So you can turn off recording audio for now while we try to figure this out.  Your ffmpeg is pretty old, maybe it doesn't have the AAC encoder built into it.  Still, we shouldn't crash, so I will have to fix that.

shaun-ba, 2017-10-23:
Hi,

I have updated ffmpeg to 3.4, i turned off audio in ZM and also in the stream. Rexd666 also suggested to set ffmpeg as the source instead of remote.

After doing these things it started to write the file which is good.

However it will not play or download through the UI it just gives a 404. The file is there however so I'm not sure why, any ideas on this part?

digital-gnome, 2017-10-23:
> However it will not play or download through the UI it just gives a 404. The file is there however 

I'd double check that the web UI is looking in exactly the right directory.  I suspect php.ini not having the correct timezone but as I recall that gave a different error.  Still, a place to start.

shaun-ba, 2017-10-23:
@digital-gnome  that was it! System time was UTC but php was Europe/London. Changed this to UTC and it now works!

Not sure why the trace about codec happened, maybe that was audio related. Now i can easily view events i can debug that more

digital-gnome, 2017-10-23:
Awesome.  That's bit me several times.  

I believe the audio part is a bit picky.  I have a camera that works fine with aac but chokes on g711a.

shaun-ba, 2017-10-24:
I'm not even sure what audio my camera uses, will have to research as there
is nothing in the UI to change etc.

It would be a shame if ZM didn't support more than a single type as I'm not
replacing these cameras!

connortechnology, 2017-10-24:
Both AAC and non-AAC should work if your ffmpeg supports AAC encoding.  They do on my systems.

shaun-ba, 2017-10-25:
@connortechnology How can i debug this? I've enabled audio with the tick box option in storage tab. Stream works fine but when an alarm triggers this happens:

2017-10-25 06:55:46.366418 | zmc_m2 |   | 29312 | INF | addr2line -e /usr/bin/zmc 0x5609aa3ed0e8 0x7f9698389390 0x7f9696c34f3c 0x7f9696c5142f 0x5609aa40f844 0x5609aa45bc07 0x5609aa3a73dc 0x5609aa37bd3e 0x7f96941aa830 0x5609aa37ce09 | zm_signal.cpp | 110
-- | -- | -- | -- | -- | -- | -- | --
2017-10-25 06:55:46.365926 | zmc_m2 |   | 29312 | INF | Backtrace complete, please execute the following command for more information | zm_signal.cpp | 109
2017-10-25 06:55:46.365424 | zmc_m2 |   | 29312 | ERR | Backtrace 9: /usr/bin/zmc(_start+0x29) [0x5609aa37ce09] | zm_signal.cpp | 102
2017-10-25 06:55:46.364884 | zmc_m2 |   | 29312 | ERR | Backtrace 8: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f96941aa830] | zm_signal.cpp | 102
2017-10-25 06:55:46.364380 | zmc_m2 |   | 29312 | ERR | Backtrace 7: /usr/bin/zmc(main+0x77e) [0x5609aa37bd3e] | zm_signal.cpp | 102
2017-10-25 06:55:46.363873 | zmc_m2 |   | 29312 | ERR | Backtrace 6: /usr/bin/zmc(_ZN7Monitor7CaptureEv+0x38c) [0x5609aa3a73dc] | zm_signal.cpp | 102
2017-10-25 06:55:46.363114 | zmc_m2 |   | 29312 | ERR | Backtrace 5: /usr/bin/zmc(_ZN12FfmpegCamera16CaptureAndRecordER5Image7timevalPc+0x6e7) [0x5609aa45bc07] | zm_signal.cpp | 102
2017-10-25 06:55:46.362129 | zmc_m2 |   | 29312 | ERR | Backtrace 4: /usr/bin/zmc(_ZN10VideoStoreD1Ev+0x544) [0x5609aa40f844] | zm_signal.cpp | 102
2017-10-25 06:55:46.361608 | zmc_m2 |   | 29312 | ERR | Backtrace 3: /usr/lib/x86_64-linux-gnu/libavformat-ffmpeg.so.56(av_write_trailer+0x28f) [0x7f9696c5142f] | zm_signal.cpp | 102
2017-10-25 06:55:46.361126 | zmc_m2 |   | 29312 | ERR | Backtrace 2: /usr/lib/x86_64-linux-gnu/libavformat-ffmpeg.so.56(+0xd3f3c) [0x7f9696c34f3c] | zm_signal.cpp | 102
2017-10-25 06:55:46.360625 | zmc_m2 |   | 29312 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f9698389390] | zm_signal.cpp | 102
2017-10-25 06:55:46.360125 | zmc_m2 |   | 29312 | ERR | Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x78) [0x5609aa3ed0e8] | zm_signal.cpp | 102
2017-10-25 06:55:46.359228 | zmc_m2 |   | 29312 | ERR | Signal address is 0x18, from 0x7f9696c34f3c | zm_signal.cpp | 81
2017-10-25 06:55:46.358710 | zmc_m2 |   | 29312 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
2017-10-25 06:55:46.358172 | zmc_m2 |   | 29312 | ERR | Error occurred when writing out file header to /var/cache/zoneminder/events/2/17/10/25/06/55/46/583-video.mp4: Invalid argument | zm_videostore.cpp | 325
2017-10-25 06:55:46.357347 | zmc_m2 |   | 29312 | WAR | Unable to set movflags to frag_custom+dash+delay_moov | zm_videostore.cpp | 317
2017-10-25 06:55:46.354534 | zmc_m2 |   | 29312 | INF | Opening video storage stream /var/cache/zoneminder/events/2/17/10/25/06/55/46/583-video.mp4 format: mp4 | zm_videostore.cpp | 54
2017-10-25 06:55:46.354218 | zma_m2 |   | 29097 | INF | Hallway: 1298 - Opening new event 583, alarm start | zm_monitor.cpp | 1514
2017-10-25 06:55:46.352868 | zma_m2 |   | 29097 | INF | Hallway: 1298 - Gone into alarm state | zm_monitor.cpp | 1471
2017-10-25 06:55:46.350609 | zma_m2 |   | 29097 | WAR | Signal: Reacquired

digital-gnome, 2017-10-26:
I think this is the important bit.  

> 10/25/17 22:48:47.884650 zmc_m6[12309].DB3-zm_videostore.cpp/202 [Have audio stream]
> 10/25/17 22:48:47.884667 zmc_m6[12309].DB2-zm_videostore.cpp/216 [Got something other than AAC (Audio: pcm_mulaw, 16000 Hz, 1 channels, s16, 128 kb/s)]
> 10/25/17 22:48:47.884737 zmc_m6[12309].DB2-zm_videostore.cpp/466 [Have audio out codec]
> 10/25/17 22:48:47.884782 zmc_m6[12309].DB2-zm_videostore.cpp/477 [Have audio_out_ctx]
> 10/25/17 22:48:47.884791 zmc_m6[12309].DB3-zm_videostore.cpp/498 [Sample rate is good]
> 10/25/17 22:48:47.884798 zmc_m6[12309].DB2-zm_ffmpeg.cpp/343 [Not fltp]
> 10/25/17 22:48:47.884804 zmc_m6[12309].DB3-zm_videostore.cpp/510 [Encoder does not support sample format s16, setting to FLTP]
> [aac @ 0x5587bdd24160] Too many bits per frame requested, clamping to max
> 10/25/17 22:48:47.886566 zmc_m6[12309].DB1-zm_videostore.cpp/547 [Audio out bit_rate (96000) sample_rate(16000) channels(1) fmt(8) layout(0) frame_size(1024)]
> 10/25/17 22:48:47.886639 zmc_m6[12309].DB1-zm_videostore.cpp/574 [Bad channel layout. Need to set it to mono (4).]
> [AVAudioResampleContext @ 0x5587bdd48520] audio_convert: found function: s16  to fltp (C)
> [AVAudioResampleContext @ 0x5587bdd48520] audio_convert: found function: s16  to fltp (SSE2)
> 10/25/17 22:48:47.886752 zmc_m6[12309].DB1-zm_ffmpeg.cpp/265 [Dumping stream index i(0) index(0)]
> 10/25/17 22:48:47.886772 zmc_m6[12309].DB1-zm_ffmpeg.cpp/271 [    Stream #0:0]
> 10/25/17 22:48:47.886778 zmc_m6[12309].DB1-zm_ffmpeg.cpp/279 [, 0, 1/90000]
> 10/25/17 22:48:47.886784 zmc_m6[12309].DB1-zm_ffmpeg.cpp/280 [: Video: h264 (High), 1 reference frame, yuvj420p(pc, bt709, left), 640x480 [SAR 4:3 DAR 16:9], 1/30, q=2-31]
> 10/25/17 22:48:47.886790 zmc_m6[12309].DB3-zm_ffmpeg.cpp/300 [
> ]
> 10/25/17 22:48:47.886797 zmc_m6[12309].DB1-zm_ffmpeg.cpp/238 [90k stream tb numerator , ]
> 10/25/17 22:48:47.886809 zmc_m6[12309].DB1-zm_ffmpeg.cpp/236 [30 codec time base:]
> 10/25/17 22:48:47.886815 zmc_m6[12309].DB1-zm_ffmpeg.cpp/330 [
> ]
> 10/25/17 22:48:47.886831 zmc_m6[12309].DB1-zm_ffmpeg.cpp/265 [Dumping stream index i(1) index(0)]
> 10/25/17 22:48:47.886842 zmc_m6[12309].DB1-zm_ffmpeg.cpp/271 [    Stream #0:1]
> 10/25/17 22:48:47.886848 zmc_m6[12309].DB1-zm_ffmpeg.cpp/279 [, 0, 0/0]
> 10/25/17 22:48:47.886854 zmc_m6[12309].DB1-zm_ffmpeg.cpp/280 [: Audio: aac, 0 channels, 128 kb/s]
> 10/25/17 22:48:47.886859 zmc_m6[12309].DB1-zm_ffmpeg.cpp/330 [
> ]
> [mp4 @ 0x5587bdceff40] sample rate not set
> 10/25/17 22:48:47.886879 zmc_m6[12309].WAR-zm_videostore.cpp/317 [Unable to set movflags to frag_custom+dash+delay_moov]
> [mp4 @ 0x5587bdceff40] sample rate not set
> 10/25/17 22:48:47.886930 zmc_m6[12309].ERR-zm_videostore.cpp/325 [Error occurred when writing out file header to /mnt/eventdrv/zoneminder/events/6/17/10/25/22/48/45/14758-video.mp4: Invalid argument
> ]
> 

Using ffmpeg from 16.04 repos which is 2.8.11-0ubuntu0.16.04.1.  If I just run ffmpeg from the command line against the stream it gives 

> The encoder 'aac' is experimental but experimental codecs are not enabled, add '-strict -2' if you want to use it.

Add -strict -2 and it runs and makes a usable video file.  So as far as I can tell it should work.

connortechnology, 2017-10-26:
I don't think that's it.  We set that flag internally.

The sample rate not being set is a problem.  Without that, the codec doesn't know what to do.  Which is funny because we make a point of setting it, and we print it out above...

digital-gnome, 2017-10-26:
> [: Audio: aac, 0 channels, 128 kb/s]

Is that 0 channels part of the problem?

connortechnology, 2017-10-26:
Could be but we check for that and set it to 1 in the output. Ffmpeg can be very finicky. Not much documentation either.

connortechnology, 2017-10-26:
I'm reviewing a lot of this code, although more to the video stream today. Maybe I'll spot something.

connortechnology, 2018-01-01:
Is this still an issue?

shaun-ba, 2018-01-01:
Have you made changes to introduce support that I should test?

shaun-ba, 2018-09-09:
@connortechnology So I've tried capturing from my camera using my laptop and get this result, can you spot anything? All i can see is that the resulting output doesn't have an audio stream.

`[rtsp @ 0x56538ade9900] max delay reached. need to consume packet
[rtsp @ 0x56538ade9900] RTP: missed 88 packets
[h264 @ 0x56538aded4a0] error while decoding MB 93 23, bytestream -19
[h264 @ 0x56538aded4a0] concealing 5356 DC, 5356 AC, 5356 MV errors in I frame
Guessed Channel Layout for Input Stream #0.1 : mono
Input #0, rtsp, from 'rtsp://admin:Saphia122014@192.168.1.154//Streaming/Channels/1':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], 16 fps, 25 tbr, 90k tbn, 32 tbc
    Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, s16, 64 kb/s
File 'output.mp4' already exists. Overwrite ? [y/N] y
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[h264 @ 0x56538afe5d40] error while decoding MB 93 23, bytestream -19
[h264 @ 0x56538afe5d40] concealing 5356 DC, 5356 AC, 5356 MV errors in I frame
[libx264 @ 0x56538ae4d320] using SAR=1/1
[libx264 @ 0x56538ae4d320] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
[libx264 @ 0x56538ae4d320] profile High, level 4.0
[libx264 @ 0x56538ae4d320] 264 - core 152 r2854 e9a5903 - H.264/MPEG-4 AVC codec - Copyleft 2003-2017 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=12 lookahead_threads=2 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=16 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=crf mbtree=1 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'output.mp4':
  Metadata:
`

shaun-ba, 2018-09-09:
@connortechnology I found something out, i added "-v verbose" to the above and got this:

```[mp4 @ 0x564a60728ac0] Could not find tag for codec adpcm_g726le in stream #1, codec not currently supported in container```

So it turns out this codec isn't supported with MP4. So i added "-c:a aac" and it works (command line ffmpeg).

So the question is do you not have handling for this already in Zoneminder? Can I add this command to the ffmpeg options?

connortechnology, 2018-09-09:
Yeah MP4 can only store AAC.  The debug logs you posted show ZM detecting that it's not AAC, at which point it sets up an encoder.  

Most cameras don't send AAC.  I have 1 that does aac, and 4 that don't.  This code works great for me... not sure why it isn't working for you.  I'll have to look closer. The only thing I can think of is that the encoder is failing to setup properly but I would expect more errors in the logs.

connortechnology, 2018-09-09:
Good news, I've been able to reproduce it.  Should be able to fix it today.  Hurray!

shaun-ba, 2018-09-09:
Ah so good!

connortechnology, 2018-09-09:
I believe I have it fixed.  Building packages now.

connortechnology, 2018-09-09:
Apparently it still crashes, just in a different spot.

shaun-ba, 2018-09-11:
@connortechnology So more work needed I take it?

connortechnology, 2018-09-11:
Oh I think I fixed that crash too.  I think it's solid.  You will have to let me know though.

shaun-ba, 2018-09-11:
Ok I will test now, I'm still stuck on an older version as I think I have
lots of crashing issues on the newer ones, but will clone and give it a try.

Thanks!

shaun-ba, 2018-09-20:
@connortechnology I've got around to testing, upgraded to 1.32.0 and hit record on the monitor with audio stream. No errors like before but I am unable to view events, back to getting these errors as I always seen to get when upgrading:

```bash

2018-09-20 09:04:29 | web_js |   | 570 | ERR | getCmdResponse stream error: Socket /var/run/zm/zms-915680s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors() | ?view=event |  
-- | -- | -- | -- | -- | -- | -- | --
2018-09-20 09:04:29 | web_php |   | 570 | ERR | Socket /var/run/zm/zms-915680s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. | /usr/share/zoneminder/www/includes/functions.php | 2042
2018-09-20 09:04:29 | web_js |   | 570 | ERR | Uncaught TypeError: Cannot read property 'getElement' of null | http://192.168.1.104/zm/cache/skins_classic_views_js_event-base-1537401771.js

```

connortechnology, 2018-09-20:
Just pushed a couple of fixes for those files.  I think we will need debug logs for zmc and zms.

shaun-ba, 2018-09-20:
I'm pretty sure this is related to apache and all those forum threads which is why I stayed on 1.31.44 until it was fixed, i think it broke in 1.31.46 ?

Did you push a new PPA version with the changes? unfortunately i'm using PPA not building from source. If not and they are apache changes let me know and i can take them from source

shaun-ba, 2018-09-20:
**This issue is resolved (audio codecs) - Thank you!**

Moving topic switch to new issue.

commit d945e2ba (2018-09-09): Fix context allocation on ffmpeg 2.8.  Fixes #41 on connorTechnology fork.
closed 2018-09-20 by shaun-ba