connortechnology/ZoneMinder #42: Unable to download MP4

shaun-ba, 2017-10-23:
Upon clicking an event, then "download mp4", the browser displays:

> This 192.168.1.x page can’t be found
> 
> No web page was found for the web address: http://192.168.1.x/zm/index.php?view=view_video&eid=4
> HTTP ERROR 404

MalakymR, 2017-10-23:
http://.../index.php?view=view_video&eid=215067 shows fine here (friday's commit 422b3d0c3349cfbba3764b23d2a5b6f537cfd6c1)

Just moved up to 7ccf06ec3ecb7bd06910cc3985257f0183ef448e and no changes here.

I would suspect its something to do with your issue #41 although the 404 is odd if you're seeing other pages with index.php

shaun-ba, 2017-10-23:
Yes, now i've logged this we know its because the file doesn't exist.

Why if an event fails in such a manner, does it record the event in the database, and reference a file that doesn't exist?

shaun-ba, 2017-10-25:
This has been resolved by matching SYSTEM timezone with PHP timezone.

closed 2017-10-25 by shaun-ba