connortechnology/ZoneMinder #44: Induvidual storage areas are not considered in filters

MalakymR, 2017-10-25:
Namely the "disk percent" filter.

The filter was set at 80% but storage crept up on 2 boxes over this number. When looking at the filter results no events are returned.
Load: 0.78 DB:15/151	Storage: cctv-test: 99%, Default: 22% /dev/shm: 57% 

Since there was only one storage area setup on these machines, I simply mounted the NFS share under /var/lib/zoneminder/events so Default now shows as the same as cctv-test.
Load: 0.67 DB:15/151	Storage: cctv-test: 99%, Default: 99% /dev/shm: 57% 

From here, events show up in the filter ready for deletion.

Updated to 0fc6852065d56c4f8432963b026365ee1e0c8954 to confirm the issue still exists.

MalakymR, 2017-10-27:
I've missed the new filter option "Storage Area"

Wouldn't it be better if nothing was selected here, it would default to everything? Goes against least surprise otherwise.

juanmoura, 2018-04-14:
I've same problem. But i have default:50% StorageA: 95% StorageB: 96%

And the Filter just work if the default is 95%.   It's ignoring the other storage areas.