connortechnology/ZoneMinder #48: Adding monitor through API fails with SQL error

Toasty27, 2017-11-07:
Running 1.31.12~20171105100058-xenial from the PPA, Ubuntu 16.04

When I query the API like so:
```
curl -XPOST http://localhost/zm/api/monitors.json -d "Monitor[Id]=157\
&Monitor[Name]=A1.1\
&Monitor[Type]=Ffmpeg\
&Monitor[Function]=Monitor\
&Monitor[Method]=rtpRtsp\
&Monitor[Path]="<redacted>"\
&Monitor[Width]=1280\
&Monitor[Height]=720\
&Monitor[Colours]=3\
&Monitor[MaxFPS]=30.00\
&Monitor[SaveJPEGS]=0\
&Monitor[VideoWriter]=2"
```

I get the following error:

```
{"success":false,"data":{"name":"An Internal Error Has Occurred.","message":"An Internal Error Has Occurred.","url":"\/zm\/api\/monitors.json","exception":{"class":"PDOException","code":"01000","message":"SQLSTATE[01000]: Warning: 1265 Data truncated for column 'Orientation' at row 1"}}}
```

This remains true even if I add `&Monitor[Orientation]='0'` to the query (no change with double quotes). I double checked the zm_create.sql file that came with the package, and everything looks correct for the Monitors table.
Specifically, it shows 
```
`Orientation` enum('0','90','180','270','hori','vert') NOT NULL default '0',
```

Adding a monitor through the web interface works as expected. Querying the API to get the monitor's settings also works. Copying those settings and using them to re-add the monitor via `curl` (after removing through the web interface) fails with the aforementioned error.

Because of that, I'm assuming the error lies somewhere in the API code, but I'm too unfamiliar with it to pinpoint where the issue lies.

I'd also like to point out that this wasn't a problem roughly one to two months ago when I last tried this (ubuntu 16.04, using the ppa, not 100% sure which version)

Toasty27, 2017-11-07:
Also tried with `&Monitor[Orientation]='1'` and `&Monitor[Orientation]='vert'`, but I'm still getting the same error.

Toasty27, 2017-11-07:
It seems as though `&Monitor[Orientation]=1` works, so the API must be using the enum's index directly. Not sure exactly why the value needed to be without quotes of any kind.

Either way, I'm going to leave this open as the API call should still be using the default value (or not providing a value at all? The table has a default value set already), which isn't happening for whatever reason.

MalakymR, 2017-11-07:
I'm not sure if its related - but if you currently try to add a monitor through the webui, it says the view does not exist. @connortechnology has said this is still going through some changes a few days ago.

connortechnology, 2018-01-01:
I may have resolved this with my latest commits.

commit 1b1b93f8 (2018-01-01): use CakePHP-Enum-Behavior to add support for ENUMs to the Monitor model.  This should fix #48
closed 2018-04-23 by connortechnology