connortechnology/ZoneMinder #77: ERR | No row for Monitor 6 |

shaun-ba, 2018-01-22:
When adding a new monitor I get these ERRs, this you can see was Monitor 6.

```
2018-01-22 15:06:15.061416 | web_js |   | 238 | ERR | Uncaught ReferenceError: monitorNames is not defined | http://192.168.1.104/zm/index.php | 80
-- | -- | -- | -- | -- | -- | -- | --
2018-01-22 15:06:03.595718 | web_php |   | 238 | ERR | No row for Monitor 6 | /usr/share/zoneminder/www/includes/Monitor.php | 162
2018-01-22 15:06:03.594910 | web_php |   | 238 | ERR | Unable to load Monitor record for Id=6 | /usr/share/zoneminder/www/includes/Monitor.php | 130
2018-01-22 15:05:53.210353 | web_php |   | 240 | ERR | Socket /var/run/zm/zms-410008s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.
```

connortechnology, 2018-01-22:
I have made some improvements here, let me know if it resolves your problem.

shaun-ba, 2018-01-30:
fixed in 1.31.29

closed 2018-01-30 by shaun-ba