connortechnology/ZoneMinder #78: ERR - Can't insert event

shaun-ba, 2018-01-23:
Another issue I've run into now with recording in Modect, can't insert events on 1.31.26

```
Can't insert event: Data truncated for column 'Scheme' at row 1. sql was (INSERT INTO Events ( MonitorId, StorageId, Name, StartTime, Width, Height, Cause, Notes, StateId, Orientation, Videoed, DefaultVideo, SaveJPEGs, Scheme ) values ( 1, 0, 'New Event', from_unixtime( 1516694165 ), 1920, 1080, 'Motion', 'Motion: All', 1, 1, 0, '', 0, '' ))
```

connortechnology, 2018-01-23:
I've pushed a new version which should fix it.  Please let me know.

shaun-ba, 2018-01-24:
Yes, this is now resolved

shaun-ba, 2018-01-24:
fixed!

connortechnology, 2018-02-04:
You need to re-run update 17.
mysql -u zmuser -p zm < /usr/share/zoneminder/db/zm_update-1.31.17.sql

shaun-ba, 2018-02-05:
Works, but shouldn't this also check/update the DB in all subsequent versions?

closed 2018-09-20 by shaun-ba