connortechnology/ZoneMinder #80: Filter Issue - PurgeWhenFull delete returned 0 events

shaun-ba, 2018-01-24:
So this isn't working anymore either, when i run the filter manually it returns the 100 results just fine, but the background process doesn't delete them, or even find them:

```
Checking filter PurgeWhenFull delete returned 0 events | zmfilter.pl
```

Could is be related to the new filter "Update disk space" ? Upon checking this filter i see that the match is:

```
Disk space                  is                    NULL
```

Is that right?

connortechnology, 2018-01-24:
I would need to see the SQL for your PrugeWhenFull filter.

 No it has nothing to do with the Update Disk Space filter.

I believe I have a fix for the third issue, with the escaping of NULL.

shaun-ba, 2018-01-24:
@connortechnology Could you give me instructions to provide that information? I've not touched the filter at all, it just doesn't find any matches when running by itself, so nothing is being removed.

connortechnology, 2018-01-24:
turn on debugging for zmfilter: go to options->Logging, check the LOG_Debug checkbox.  
make sure that LOG_DEBUG_TARGET is either empty or contains _zmfilter. Make sure LOG_DEBUG_FILE is empty, and LOG_LEVEL_FILE is Debug.  Then there should be lots of output in /var/log/zm/zmfilter.log

shaun-ba, 2018-01-25:
@connortechnology thanks, will remember that for the future.

Here is the log

```
01/25/18 11:56:40.751403 zmfilter[15323].INF [Checking filter PurgeWhenFull delete returned 0 events ]
01/25/18 11:56:40.753741 zmfilter[15323].DBG [Filter::Execute SQL (SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         INNER JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND ( E.DiskSpace IS NULL ) order by E.StartTime desc)]

01/25/18 11:56:40.768031 zmfilter[15323].DBG [Loaded 0 events for filter Update DiskSpace using query (SELECT E.*,
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         INNER JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND ( E.DiskSpace IS NULL ) order by E.StartTime desc)]
```

shaun-ba, 2018-01-30:
fixed in 1.31.29

commit a271f177 (2018-01-24): Fix #80 don't escape NULL value when building SQL
closed 2018-01-30 by shaun-ba