connortechnology/ZoneMinder #81: ERR | SQL-ERR 'SQLSTATE[42000]: Syntax error

shaun-ba, 2018-01-24:
Another ERR i came across in logs, thought i'd log it.

```
2018-01-24 13:36:04.324562 | web_php |   | 238 | ERR | SQL-ERR 'SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '-25, 25' at line 1', statement was 'SELECT E.*,M.Name AS MonitorName,M.DefaultScale FROM Monitors AS M INNER JOIN Events AS E on (M.Id = E.MonitorId) WHERE 1 and ( E.StartTime >= '2018-01-24 12:36:04' and E.MonitorId = '2' ) ORDER BY E.StartTime asc limit -25, 25' params:
-- | -- | -- | -- | -- | --
```

connortechnology, 2018-01-24:
The limit is a text box, you have entered -25,25 into it.  I will have to add filtering of non-numeric input.

shaun-ba, 2018-01-24:
What box? No box in the filters anywhere that i can see has -25,25 in it

shaun-ba, 2018-01-30:
fixed in 1.31.29

commit c7ecd580 (2018-01-24): Fix #81 by testing for non-numeric characters in LIMIT.  Change type of input to number, making it harder to enter non-numeric characters
closed 2018-01-30 by shaun-ba