connortechnology/ZoneMinder #83: Storage: Default: -- Negative value

shaun-ba, 2018-02-19:
v1.31.37

Storage: Default: -20%

tooltip: "-21250153547B of 97.32GB"

Why is this value negative? I'm insure what space i'm actually using, is it 20% of total?

shaun-ba, 2018-02-28:
I think this is causing me major issues now with regard to events being deleted. I have a large volume here for storage and only 30GB is being stored. 

Any idea on how to "reset" this, or fix it?

connortechnology, 2018-02-28:
No, it is used for display purposes only.  Please post your filters.

shaun-ba, 2018-03-01:
Hi,

I only have two.

PurgeWhenFull
Archived status        Unarchived only
Disk Percent            80

Update DiskSpace
Disk Space      IS      NULL

Events are only being stored for the current day, nothing older. Day + Week event counts are all the same for all cameras, this is strange right because some have 11 events and some have 200, so you would assume the older ones would be pruned first if it was a disk space issue?

connortechnology, 2018-03-01:
Ok, 1.39 fixes the second issue you just mentioned, which is the day, hour and week counts being the same and wrong.  Please update to 1.39.  I don't think it fixes the diskspace totals in the header though.  As I said before, the value is used for display purposes only. In the filters they will actually use the real disk space as reported by the php functions for doing so.

How about turning on debug and seeing the zmfilter.pl logs.

shaun-ba, 2018-03-01:
It wasn't just the counts, there were no events even when clicked, maybe that's what you mean also.

Ok I'll add this logging now and see what turns up.

closed 2018-09-20 by shaun-ba