connortechnology/ZoneMinder #87: Zoneminder background process leaking logs to foreground.

shaun-ba, 2018-03-12:
Zoneminder background process leaking logs to foreground.

Replicate: Login to server, wait a few seconds and start seeing logs spammed to foreground.

```
n@zm-vid:~$ 
Broadcast message from systemd-journald@zm-vid (Mon 2018-03-12 15:41:51 UTC):

zma_m11[539]:  [zma_m11] [Rear-sub: 6255000 - Analysing at 5.99 fps]

Broadcast message from systemd-journald@zm-vid (Mon 2018-03-12 15:41:52 UTC):

zma_m8[503]:  [zma_m8] [Front-sub: 6255000 - Analysing at 6.02 fps]
```

pliablepixels, 2018-03-24:
@connortechnology - the fixes did not solve this problem - its gotten worse (more frequent)

connortechnology, 2018-03-24:
weird. Anyways, turn off logging to syslog.  Or set it to critical instead of info..

shaun-ba, 2018-03-25:
Ok, that's a temporary workaround whilst you investigate and fix it?

shaun-ba, 2018-03-28:
@pliablepixels I've not come across this with ANY software in the past, very odd!

connortechnology, 2018-03-29:
So what seems to be happening is that when the process is told to rotate logs, it starts doing this.  So something is wrong in the log reload functions.  

Should have a fix soon.

connortechnology, 2018-03-29:
I think I found it.

Please note that the behaviour only happens after a reload, so to test, all you have do is
killall -HUP zma

pliablepixels, 2018-03-29:
so far so good

shaun-ba, 2018-05-08:
resolved!

closed 2018-05-08 by shaun-ba