connortechnology/ZoneMinder #103: ZMC Backtrace

shaun-ba, 2018-09-28:
@connortechnology Unfortunately another one, on 1.32.0 this time:

2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 7: /usr/bin/zmc(_start+0x29) [0x55c6b1cd20f9]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f5d24a74830]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 5: /usr/bin/zmc(main+0xa78) [0x55c6b1ccf948]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 4: /usr/bin/zmc(_ZN12FfmpegCamera5CloseEv+0x146) [0x55c6b1da5e86]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 3: /usr/bin/zmc(_ZN10VideoStoreD1Ev+0x7c7) [0x55c6b1d5ba07]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 2: /usr/lib/x86_64-linux-gnu/libavcodec-ffmpeg.so.56(avcodec_close+0x33) [0x7f5d26098cbf]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f5d28c5a390]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x78) [0x55c6b1d3e768]	zm_signal.cpp	102
2018-09-28 08:55:09	zmc_m3		27846	ERR	Signal address is (nil), from 0x7f5d26098cbf	zm_signal.cpp	81
2018-09-28 08:55:09	zmc_m3		27846	ERR	Got signal 11 (Segmentation fault), crashing

connortechnology, 2018-09-28:
We have fixed a couple more crashes and weirdnesses over the last two days.  Is your code up to date as of yesterday afternoon? If not, please update and try again.

connortechnology, 2018-09-28:
Oh and I will need logs.

shaun-ba, 2018-09-28:
@connortechnology  I try my best to get logs, but with the previous issue I had with codec there were none, in any log file with top level debugging on.

Does ZM support H.264+ encoding ?

I have monitors on record at 10minute intervals, and i'm getting lots of random length events, but no ERR logs, which component should I add debugging to to get logs for that ?

connortechnology, 2018-09-28:
I have no idea about h.264+ encoding.  Regardless it shouldn't cause a crash.

You say you have no err logs, but above you posted err logs.  WHich is it?  I need debug logs so I can guess where and why the crash happens. All I can see from above is that the capture stream closed and it deleted the VideoStore object and it all crashed.

shaun-ba, 2018-09-29:
@connortechnology Yes, just this one, you asked for more logs but this is all I have. I can't go back and get them if I don't have debugging on for that component, so which component is it you need them for?

I post more issues that anymore here, 75% being genuine bugs so I wouldn't treat my reports as a chore as I'm trying to aid in making the software more stable.

shaun-ba, 2018-10-01:
@connortechnology I disabled the H264+ and I've not had any event drop outs so far, I'm not sure about the scrambling of footage yet as nobody has crossed the frame, but it may be useful to know.

I actually enabled H264+ as it decreases bandwidth and filesize by 50%, apparently it's close to H265. I did this to see if the scrambling of footage was related to bandwidth, but I now thing it's zoneminder related and nothing to do with bandwidth or WiFi.

shaun-ba, 2018-10-01:
@connortechnology The scrambling of events, artefacts and such, only occur when Audio is enabled.

I have disabled audio on these cameras and I've checked 30minutes of footage with no issues so far, so I think it's safe to say that this is related to that.

connortechnology, 2018-10-11:
I just noticed that I have a dahua camera that claims to be sending h264+ with audio.  

Works perfectly for me.  :)

That last debug trace shows me something that I was expected, which is 0/0 timebase.  I had added a test for that, so I'm wondering if this is still an issue?

shaun-ba, 2018-10-11:
@connortechnology Unless you've pushed a fix since 1.32.0 for this then all of my issues are still valid. I've tried my best to post as much information and logs as possible so maybe you can correlate something between the different issues. 

If not please let me know exactly how I can help, because I've tried various debugging logs and such but nothing seems to flag up.

connortechnology, 2018-10-24:
I just fixed an issue that turned up on ubuntu 1810 .  Related to audio codec not working due to channel layout being incorrect... 

Please try latest.